persuader72/esphome-components #4: shelly_dht cannot be used for climate thermostat

alexander-vitishchenko, 2021-10-12:
Hi @persuader72 

Just wanted to thank you for your great "shelly_dallas" integration one more time! I'm now able to use Shelly 1 as a heater thermostat and it works great!

I'm not working on another type of automation - dry air in the bathroom when the humidity level exceeds a certain level. I'm trying to connect my DHT sensor and use it as part of climate thermostat for ESPHome... but get the following error message. Can you please advise if I do something wrongly, or your "shelly_dht" is not suited for this kind of appliance... as I would guess by the different structure of the config file from your tutorial? 

Thank you,
Alexander

[image]

alexander-vitishchenko, 2021-10-12:
ESP Home version: 2021.9.3
Home Assistant version: core-2021.10.4

persuader72, 2021-10-14:
Hi @alexander-vitishchenko I think I found the problem...

you have to put an ID on the temperature_sensor inside the shelly_dht device as shown in the following picture. And in the climate component, you have to refer to the ID of the temperature sensor and not to the ID of the whole shelly_dht.

This is because the shelly_dht is the union of two sensors (temperature and humidity) but the climate wants a simple sensor.

[image]

alexander-vitishchenko, 2021-10-16:
Thanks @persuader72!

True - my config has to be adjusted as you advised. I've complied and uploaded new firmware to my Shelly 1.

The final step left is to connect my DHT sensor and see if it works correctly.

alexander-vitishchenko, 2021-10-16:
@persuader72 

My current sensor is recognized as DHT22, but then no temperature and humidity readings are available. 

I'm pretty sure your firmware should work well, and the problem is that my sensor is neither DHT11 nor DHT22. Perhaps it is DHT21, or some other exotic.

I'll try to buy DHT22 sensor tomorrow (Sunday), and then connect it to Shelly 1 again. Hope it will work.

See photo attached, in case you can recognize the model by any chance.
[image]

alexander-vitishchenko, 2021-10-17:
@persuader72 just realised that buying new DHT22 sensor(s) will help me in around 6 places at home, where I have these sensors visible/accessible/replaceable

And I have about 5 of the other existing sensors that could be hardly replaced (within the floor) and the ideal would be to make them work with Shelly 1. So I’ll proceed with testing for both the existing sensors (of unknown model) and new DHT22

I know this could be a challenge… but please let me know if you by any chance know how to identify those my existing sensor models so we can see if it possible to get them connected to ESPHome. Otherwise I’ll have to break through the floor, or use separate wireless sensors with virtual climate thermostat on Home Assistant side rather than within ESPHome

alexander-vitishchenko, 2021-10-17:
UPDATE:

I've purchased DHT11 sensor for testing (which wasn't using on Sunday) - and your firmware proves to be working like a charm! My new DHT22 sensor to be delivered in few days time, and I expect it to work too.

This leaves me with the only open question what to do with my 4 existing sensors that have unidentified model

alexander-vitishchenko, 2021-10-31:
Closing the issue as the initial request has been resolved. As for unknown temperature/humidity sensor - this is a separate question and I've decided to replace them with well known DHT11 or DHT22

closed 2021-10-31 by alexander-vitishchenko