persuader72/esphome-components #6: Found no sensors

MarkoPaasila, 2022-01-05:
What could be the reason? 
```
[12:59:01][C][dallas.sensor:072]: ShellyDallasComponent:
[12:59:01][C][dallas.sensor:073]:   In Pin: GPIO3
[12:59:01][C][dallas.sensor:074]:   Out Pin: GPIO0
[12:59:01][C][dallas.sensor:075]:   Update Interval: 60.0s
[12:59:01][W][dallas.sensor:078]:   Found no sensors!
```
I'm using dev branch. Here's the relevant part from my config:
```
shelly_dallas:
  pin_a: GPIO3
  pin_b: GPIO0
    
sensor:
#  - platform: shelly_dallas
#    address: 0xCA011938282E5128
#    index: 0
#    name: "Living Room Floor"
#    accuracy_decimals: 1
#    unit_of_measurement: "°C"
#    device_class: temperature
#    state_class: measurement
#    icon: "mdi:thermometer"
```
The sensor is commented out. I've tried the sensor component using the address (which is wrong as I can't get the correct one) and the index parameter (according to [ESPHome Dallas page](https://esphome.io/components/sensor/dallas.html?highlight=dallas#id1). Neither work.

persuader72, 2022-01-05:
This Is the same procedure I followed for my sensors. First I run the component without sensors to find out the addresses of the probes. Next, I configured the sensors using the discovered addresses.

Which version of esphome are you using? I can try to understand if there is some compatibility issue.

In past, I had a similar problem but turned out to be a wire connection problem.

MarkoPaasila, 2022-01-05:
The newest esphome.  I also tried esp8266 platform latest and dev versions. Same thing. I checked that the sensor works on stock firmware with the same hardware.

persuader72, 2022-01-05:
I found this commit https://github.com/esphome/issues/issues/2578 of 3 months ago on the main esphome branch But my ShellyDallas Component  doesn't contain this patch because is based on an older version of the same component

I will try to align my component to the last esphome version but I think this will take some days.
 I will update you when I have finished.

Regards
Stefano

MarkoPaasila, 2022-03-10:
Sorry to bug you but have you managed to have a look at this?

MarkoPaasila, 2022-03-15:
Is this relevant? https://github.com/esphome/esphome/pull/3293

closed 2024-05-16 by persuader72