persuader72/esphome-components #9: Sensor broken in latest ESP Home

srnoth, 2022-03-16:
ESP Home version 2022.3.0

```
ERROR Unable to load component shelly_dht.sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/custom_components/shelly_dht/sensor.py", line 26, in <module>
    cv.Optional(CONF_TEMPERATURE): sensor.sensor_schema(UNIT_CELSIUS, ICON_THERMOMETER, 1),
TypeError: sensor_schema() takes from 0 to 1 positional arguments but 3 were given
Failed config

Platform not found: 'sensor.shelly_dht'.
```

kcinnaySte, 2022-03-17:
Hey srnoth,
please try PullRequest #10

cg089, 2022-03-25:
I can confirm #10 works

closed 2024-05-16 by persuader72