persuader72/esphome-components #11: 'uint64_to_string' was not declared in this scope

Severinius, 2022-07-06:
Unable compiling DALLAS Sensor for latest ESPHOME 2022.6.2

src/esphome/components/shelly_dallas/shelly_dallas_component.cpp: In member function 'virtual void esphome::shelly_dallas::ShellyDallasComponent::setup()':
src/esphome/components/shelly_dallas/shelly_dallas_component.cpp:42:21: error: 'uint64_to_string' was not declared in this scope
   42 |     std::string s = uint64_to_string(address);
      |                     ^~~~~~~~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_dallas_component.cpp: In member function 'virtual void esphome::shelly_dallas::ShellyDallasComponent::dump_config()':
src/esphome/components/shelly_dallas/shelly_dallas_component.cpp:82:23: error: 'uint64_to_string' was not declared in this scope
   82 |       std::string s = uint64_to_string(address);
      |                       ^~~~~~~~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_dallas_component.cpp: In member function 'const string& esphome::shelly_dallas::ShellyDallasTemperatureSensor::get_address_name()':
src/esphome/components/shelly_dallas/shelly_dallas_component.cpp:174:47: error: 'uint64_to_string' was not declared in this scope
  174 |     this->address_name_ = std::string("0x") + uint64_to_string(this->address_);
      |                                               ^~~~~~~~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_dallas_component.cpp: In member function 'bool esphome::shelly_dallas::ShellyDallasTemperatureSensor::setup_sensor()':
src/esphome/components/shelly_dallas/shelly_dallas_component.cpp:197:32: warning: 'bool esphome::shelly_dallas::ShellyDallasTemperatureSensor::read_scratch_pad()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  197 |     r = this->read_scratch_pad();
      |                                ^
src/esphome/components/shelly_dallas/shelly_dallas_component.cpp:179:22: note: declared here
  179 | bool ICACHE_RAM_ATTR ShellyDallasTemperatureSensor::read_scratch_pad() {
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_dallas_component.cpp: In member function 'virtual std::string esphome::shelly_dallas::ShellyDallasTemperatureSensor::unique_id()':
src/esphome/components/shelly_dallas/shelly_dallas_component.cpp:271:77: error: 'uint64_to_string' was not declared in this scope
  271 | std::string ShellyDallasTemperatureSensor::unique_id() { return "dallas-" + uint64_to_string(this->address_); }
      |                                                                             ^~~~~~~~~~~~~~~~
*** [/data/gastroback/.pioenvs/gastroback/src/esphome/components/shelly_dallas/shelly_dallas_component.cpp.o] Error 1

Thanks for your help!

pedroona, 2022-07-25:
Same problem here.
Using [v2022.6.2] of esphome
Did you find a solution @Severinius?

Gobbo28, 2022-07-27:
Hello all,

I am not that experienced in programming but I compared the current version of the dallas component with the shelly_dallas code and made some corrections to make it work in my system.
There are still some warnings about deprecated components but at the moment it's working.

https://github.com/Gobbo28/esphome_shelly_dallas_fork

closed 2024-05-16 by persuader72