persuader72/esphome-components #14: Can't Compile with  shelly_dallas

marcolino7, 2023-07-25:
Hi,
I am just new with HA and ESPHome. I made my first node with shelly1 and Shelly1 Addon and 1 dallas sensor.
I created forder and copied file, and yaml syntax is correct.
When I am going to compile i got this:

`INFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/corridoio1.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing corridoio1 (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- AsyncMqttClient-esphome @ 0.8.6
|-- ArduinoJson @ 6.18.5
Compiling /data/corridoio1/.pioenvs/corridoio1/src/esphome/components/shelly_dallas/shelly_dallas_component.cpp.o
Compiling /data/corridoio1/.pioenvs/corridoio1/src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp.o
In file included from src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:1:
src/esphome/components/shelly_dallas/shelly_esp_one_wire.h:4:10: fatal error: esphome/core/esphal.h: No such file or directory
    4 | #include "esphome/core/esphal.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/corridoio1/.pioenvs/corridoio1/src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp.o] Error 1
In file included from src/esphome/components/shelly_dallas/shelly_dallas_component.h:5,
                 from src/esphome/components/shelly_dallas/shelly_dallas_component.cpp:1:
src/esphome/components/shelly_dallas/shelly_esp_one_wire.h:4:10: fatal error: esphome/core/esphal.h: No such file or directory
    4 | #include "esphome/core/esphal.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/corridoio1/.pioenvs/corridoio1/src/esphome/components/shelly_dallas/shelly_dallas_component.cpp.o] Error 1
========================== [FAILED] Took 2.23 seconds ==========================`

I am using ESPHome plugin for home assistant and I have no idea on how to fix.
Many thanks in advance

Marco

marcolino7, 2023-07-25:
Hi,
using dev version I have different error

```
INFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/corridoio1.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing corridoio1 (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- AsyncMqttClient-esphome @ 0.8.6
|-- ArduinoJson @ 6.18.5
Compiling /data/corridoio1/.pioenvs/corridoio1/src/esphome/components/shelly_dallas/shelly_dallas_component.cpp.o
Compiling /data/corridoio1/.pioenvs/corridoio1/src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp.o
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'void esphome::shelly_dallas::ESPOneWire::write8(uint8_t)':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:85:42: warning: 'void esphome::shelly_dallas::ESPOneWire::write_bit(bool)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
   85 |     this->write_bit(bool((1u << i) & val));
      |                                          ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:45:26: note: declared here
   45 | void HOT ICACHE_RAM_ATTR ESPOneWire::write_bit(bool bit) {
      |                          ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'void esphome::shelly_dallas::ESPOneWire::write64(uint64_t)':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:91:44: warning: 'void esphome::shelly_dallas::ESPOneWire::write_bit(bool)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
   91 |     this->write_bit(bool((1ULL << i) & val));
      |                                            ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:45:26: note: declared here
   45 | void HOT ICACHE_RAM_ATTR ESPOneWire::write_bit(bool bit) {
      |                          ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'uint8_t esphome::shelly_dallas::ESPOneWire::read8()':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:98:36: warning: 'bool esphome::shelly_dallas::ESPOneWire::read_bit()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
   98 |     ret |= (uint8_t(this->read_bit()) << i);
      |                                    ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:66:26: note: declared here
   66 | bool HOT ICACHE_RAM_ATTR ESPOneWire::read_bit() {
      |                          ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'uint64_t esphome::shelly_dallas::ESPOneWire::read64()':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:105:37: warning: 'bool esphome::shelly_dallas::ESPOneWire::read_bit()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  105 |     ret |= (uint64_t(this->read_bit()) << i);
      |                                     ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:66:26: note: declared here
   66 | bool HOT ICACHE_RAM_ATTR ESPOneWire::read_bit() {
      |                          ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'void esphome::shelly_dallas::ESPOneWire::select(uint64_t)':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:110:35: warning: 'void esphome::shelly_dallas::ESPOneWire::write8(uint8_t)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  110 |   this->write8(ONE_WIRE_ROM_SELECT);
      |                                   ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:83:22: note: declared here
   83 | void ICACHE_RAM_ATTR ESPOneWire::write8(uint8_t val) {
      |                      ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:111:24: warning: 'void esphome::shelly_dallas::ESPOneWire::write64(uint64_t)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  111 |   this->write64(address);
      |                        ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:89:22: note: declared here
   89 | void ICACHE_RAM_ATTR ESPOneWire::write64(uint64_t val) {
      |                      ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'uint64_t esphome::shelly_dallas::ESPOneWire::search()':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:124:20: warning: 'bool esphome::shelly_dallas::ESPOneWire::reset()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  124 |   if (!this->reset()) {
      |                    ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:19:26: note: declared here
   19 | bool HOT ICACHE_RAM_ATTR ESPOneWire::reset() {
      |                          ^~~~~~~~~~
[... 58 lines cut ...]
   42 |     std::string s = uint64_to_string(address);
[... cut ...]
src/esphome/components/shelly_dallas/shelly_dallas_component.cpp:271:77: error: 'uint64_to_string' was not declared in this scope
  271 | std::string ShellyDallasTemperatureSensor::unique_id() { return "dallas-" + uint64_to_string(this->address_); }
      |                                                                             ^~~~~~~~~~~~~~~~
Compiling /data/corridoio1/.pioenvs/corridoio1/src/main.cpp.o
*** [/data/corridoio1/.pioenvs/corridoio1/src/esphome/components/shelly_dallas/shelly_dallas_component.cpp.o] Error 1
========================== [FAILED] Took 7.17 seconds ==========================
```

dcorwin8222, 2023-08-18:
I am getting similar issues:

`
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'void esphome::shelly_dallas::ESPOneWire::write8(uint8_t)':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:85:42: warning: 'void esphome::shelly_dallas::ESPOneWire::write_bit(bool)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
   85 |     this->write_bit(bool((1u << i) & val));
      |                                          ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:45:26: note: declared here
   45 | void HOT ICACHE_RAM_ATTR ESPOneWire::write_bit(bool bit) {
      |                          ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'void esphome::shelly_dallas::ESPOneWire::write64(uint64_t)':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:91:44: warning: 'void esphome::shelly_dallas::ESPOneWire::write_bit(bool)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
   91 |     this->write_bit(bool((1ULL << i) & val));
      |                                            ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:45:26: note: declared here
   45 | void HOT ICACHE_RAM_ATTR ESPOneWire::write_bit(bool bit) {
      |                          ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'uint8_t esphome::shelly_dallas::ESPOneWire::read8()':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:98:36: warning: 'bool esphome::shelly_dallas::ESPOneWire::read_bit()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
   98 |     ret |= (uint8_t(this->read_bit()) << i);
      |                                    ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:66:26: note: declared here
   66 | bool HOT ICACHE_RAM_ATTR ESPOneWire::read_bit() {
      |                          ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'uint64_t esphome::shelly_dallas::ESPOneWire::read64()':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:105:37: warning: 'bool esphome::shelly_dallas::ESPOneWire::read_bit()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  105 |     ret |= (uint64_t(this->read_bit()) << i);
      |                                     ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:66:26: note: declared here
   66 | bool HOT ICACHE_RAM_ATTR ESPOneWire::read_bit() {
      |                          ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'void esphome::shelly_dallas::ESPOneWire::select(uint64_t)':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:110:35: warning: 'void esphome::shelly_dallas::ESPOneWire::write8(uint8_t)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  110 |   this->write8(ONE_WIRE_ROM_SELECT);
      |                                   ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:83:22: note: declared here
   83 | void ICACHE_RAM_ATTR ESPOneWire::write8(uint8_t val) {
      |                      ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:111:24: warning: 'void esphome::shelly_dallas::ESPOneWire::write64(uint64_t)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  111 |   this->write64(address);
      |                        ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:89:22: note: declared here
   89 | void ICACHE_RAM_ATTR ESPOneWire::write64(uint64_t val) {
      |                      ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'uint64_t esphome::shelly_dallas::ESPOneWire::search()':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:124:20: warning: 'bool esphome::shelly_dallas::ESPOneWire::reset()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  124 |   if (!this->reset()) {
      |                    ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:19:26: note: declared here
   19 | bool HOT ICACHE_RAM_ATTR ESPOneWire::reset() {
      |                          ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:126:24: warning: 'void esphome::shelly_dallas::ESPOneWire::reset_search()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  126 |     this->reset_search();
      |                        ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:113:22: note: declared here
  113 | void ICACHE_RAM_ATTR ESPOneWire::reset_search() {
      |                      ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:138:35: warning: 'void esphome::shelly_dallas::ESPOneWire::write8(uint8_t)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  138 |   this->write8(ONE_WIRE_ROM_SEARCH);
      |                                   ^
[... cut ...]
src/esphome/components/shelly_dallas/shelly_dallas_component.cpp: In member function 'virtual std::string esphome::shelly_dallas::ShellyDallasTemperatureSensor::unique_id()':
src/esphome/components/shelly_dallas/shelly_dallas_component.cpp:271:77: error: 'uint64_to_string' was not declared in this scope
  271 | std::string ShellyDallasTemperatureSensor::unique_id() { return "dallas-" + uint64_to_string(this->address_); }
      |                                                                             ^~~~~~~~~~~~~~~~
*** [/data/water-heater/.pioenvs/water-heater/src/esphome/components/shelly_dallas/shelly_dallas_component.cpp.o] Error 1`

persuader72, 2023-08-19:
Hi made  the fixes and now is compiling on esphome 2023.8.1
Beacuse my dallas is currently busy and I can use it to test the firmeare can you test it please?

dcorwin8222, 2023-08-19:
I was able to install the component and get my sensor IDs. However when I put them in the rest of my config it errors out.

I attached 2 txt files one w/ the error and one w/ my esphome config
[esphome_wh_config.txt](https://github.com/persuader72/esphome-components/files/12385625/esphome_wh_config.txt)
[esphome_wh_error.txt](https://github.com/persuader72/esphome-components/files/12385626/esphome_wh_error.txt)

dcorwin8222, 2023-08-19:
I got it working:
- Flashed Shelly 1 w/ just the shelly_dallas config w/ pins. That spit out the addresses of the dallas sensor in the logged data
- I had to convert the addresses given to hex and use the hex addresses in the config
- I had to flash the Shelly 1 w/ a config that was bare minimal then I was able to flash back the config file w/ the dallas sensor settings.

persuader72, 2023-08-19:
Maybe the alternative function of uint64_to_string take to much memory.
Have you the ability to test the [dev branch](https://github.com/persuader72/esphome-components/tree/dev). I merged a [pull request](https://github.com/persuader72/esphome-components/pull/7) form  @jowgn and I think that his solution take few memory.

marcolino7, 2023-08-23:
Hi @persuader72 ,
I have tried dev branch. It compile but I get some Warning:

```
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'void esphome::shelly_dallas::ESPOneWire::write8(uint8_t)':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:85:42: warning: 'void esphome::shelly_dallas::ESPOneWire::write_bit(bool)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
   85 |     this->write_bit(bool((1u << i) & val));
      |                                          ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:45:26: note: declared here
   45 | void HOT ICACHE_RAM_ATTR ESPOneWire::write_bit(bool bit) {
      |                          ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'void esphome::shelly_dallas::ESPOneWire::write64(uint64_t)':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:91:44: warning: 'void esphome::shelly_dallas::ESPOneWire::write_bit(bool)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
   91 |     this->write_bit(bool((1ULL << i) & val));
      |                                            ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:45:26: note: declared here
   45 | void HOT ICACHE_RAM_ATTR ESPOneWire::write_bit(bool bit) {
      |                          ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'uint8_t esphome::shelly_dallas::ESPOneWire::read8()':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:98:36: warning: 'bool esphome::shelly_dallas::ESPOneWire::read_bit()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
   98 |     ret |= (uint8_t(this->read_bit()) << i);
      |                                    ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:66:26: note: declared here
   66 | bool HOT ICACHE_RAM_ATTR ESPOneWire::read_bit() {
      |                          ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'uint64_t esphome::shelly_dallas::ESPOneWire::read64()':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:105:37: warning: 'bool esphome::shelly_dallas::ESPOneWire::read_bit()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  105 |     ret |= (uint64_t(this->read_bit()) << i);
      |                                     ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:66:26: note: declared here
   66 | bool HOT ICACHE_RAM_ATTR ESPOneWire::read_bit() {
      |                          ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'void esphome::shelly_dallas::ESPOneWire::select(uint64_t)':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:110:35: warning: 'void esphome::shelly_dallas::ESPOneWire::write8(uint8_t)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  110 |   this->write8(ONE_WIRE_ROM_SELECT);
      |                                   ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:83:22: note: declared here
   83 | void ICACHE_RAM_ATTR ESPOneWire::write8(uint8_t val) {
      |                      ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:111:24: warning: 'void esphome::shelly_dallas::ESPOneWire::write64(uint64_t)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  111 |   this->write64(address);
      |                        ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:89:22: note: declared here
   89 | void ICACHE_RAM_ATTR ESPOneWire::write64(uint64_t val) {
      |                      ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp: In member function 'uint64_t esphome::shelly_dallas::ESPOneWire::search()':
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:124:20: warning: 'bool esphome::shelly_dallas::ESPOneWire::reset()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  124 |   if (!this->reset()) {
      |                    ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:19:26: note: declared here
   19 | bool HOT ICACHE_RAM_ATTR ESPOneWire::reset() {
      |                          ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:126:24: warning: 'void esphome::shelly_dallas::ESPOneWire::reset_search()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  126 |     this->reset_search();
      |                        ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:113:22: note: declared here
  113 | void ICACHE_RAM_ATTR ESPOneWire::reset_search() {
      |                      ^~~~~~~~~~
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:138:35: warning: 'void esphome::shelly_dallas::ESPOneWire::write8(uint8_t)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  138 |   this->write8(ONE_WIRE_ROM_SEARCH);
      |                                   ^
[... cut ...]
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:215:20: warning: 'void esphome::shelly_dallas::ESPOneWire::write8(uint8_t)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  215 |   this->write8(0xCC);  // skip ROM
      |                    ^
src/esphome/components/shelly_dallas/shelly_esp_one_wire.cpp:83:22: note: declared here
   83 | void ICACHE_RAM_ATTR ESPOneWire::write8(uint8_t val) {
      |                      ^~~~~~~~~~
```

but it work fine. Thanks

persuader72, 2024-05-16:
Update the shelly_dallas component to esphome 24.2.1

closed 2024-05-16 by persuader72