persuader72/esphome-components #16: Sensor addresses too long?

ldeffenb, 2023-10-02:
I have a Shelly 1PM converted to ESPhome.  The 1PM has 3 attached temperature sensors that were working fine under Shelly.  I've added these custom components to ESPhome and built with the following bit in my node's .YAML:
```
shelly_dallas:  # https://github.com/persuader72/esphome-components
  pin_a: GPIO3
  pin_b: GPIO0
```
This resulted in the following output when it downloaded and ran (not sure why dallas.sensor, I would have expected shelly_dallas.sensor):
```
[11:51:51][D][dallas.sensor:089]:   Found sensors:
[11:51:51][D][dallas.sensor:092]:     0x1082014344088452904
[11:51:51][D][dallas.sensor:092]:     0x14628842022955515688
[11:51:51][D][dallas.sensor:092]:     0x2451108630715039528
```
I then added the following bit for these sensors:
```
sensor:
  - platform: shelly_dallas
    address: "0x1082014344088452904"
    name: "Temperature #1"
  - platform: shelly_dallas
    address: "0x14628842022955515688"
    name: "Temperature #2"
  - platform: shelly_dallas
    address: "0x2451108630715039528"
    name: "Temperature #3"
```
But I now get the following warnings on the build, and the subsequent output at runtime.   Needless to say, the sensors are not coming online and are displaying "nan" via MQTT and "NA" in the web view.
```
Compiling .pioenvs/fridge1pm/src/main.cpp.o
src/main.cpp:392:108: warning: integer constant is too large for its type
  392 |   shelly_dallas_shellydallastemperaturesensor = shelly_dallas_shellydallascomponent->get_sensor_by_address(0x1082014344088452904, 12);
      |                                                                                                            ^~~~~~~~~~~~~~~~~~~~~
src/main.cpp:411:110: warning: integer constant is too large for its type
  411 |   shelly_dallas_shellydallastemperaturesensor_2 = shelly_dallas_shellydallascomponent->get_sensor_by_address(0x14628842022955515688, 12);
      |                                                                                                              ^~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:430:110: warning: integer constant is too large for its type
  430 |   shelly_dallas_shellydallastemperaturesensor_3 = shelly_dallas_shellydallascomponent->get_sensor_by_address(0x2451108630715039528, 12);
      | 
```
Note that the addresses shown here do NOT match the ones configured and warned about.   I'm guessing an integer truncation between the configuration and the uint64 address.
```
[12:10:45][C][dallas.sensor:097]:   Device 'Temperature #1'
[12:10:45][C][dallas.sensor:097]:     State Class: ''
[12:10:45][C][dallas.sensor:097]:     Unit of Measurement: ''
[12:10:45][C][dallas.sensor:097]:     Accuracy Decimals: 0
[12:10:45][C][dallas.sensor:105]:     Address: 0x2311529960516692228
[12:10:45][C][dallas.sensor:106]:     Resolution: 12
[12:10:45][C][dallas.sensor:097]:   Device 'Temperature #2'
[12:10:45][C][dallas.sensor:097]:     State Class: ''
[12:10:45][C][dallas.sensor:097]:     Unit of Measurement: ''
[12:10:45][C][dallas.sensor:097]:     Accuracy Decimals: 0
[12:10:45][C][dallas.sensor:105]:     Address: 0x9818412514169411208
[12:10:45][C][dallas.sensor:106]:     Resolution: 12
[12:10:45][C][dallas.sensor:097]:   Device 'Temperature #3'
[12:10:45][C][dallas.sensor:097]:     State Class: ''
[12:10:45][C][dallas.sensor:097]:     Unit of Measurement: ''
[12:10:45][C][dallas.sensor:097]:     Accuracy Decimals: 0
[12:10:45][C][dallas.sensor:105]:     Address: 0x1227339780526937384
[12:10:45][C][dallas.sensor:106]:     Resolution: 12
```
Any helpful suggestions would be appreciated.

ldeffenb, 2023-10-02:
Oh, and when the devices poll, the following shows up in the logs:
```
[12:39:37][W][dallas.sensor:156]: 'Temperature #1' - Scratch pad checksum invalid!
[12:39:38][D][sensor:093]: 'Temperature #1': Sending state nan °C with 0 decimals of accuracy
[12:39:38][W][dallas.sensor:156]: 'Temperature #2' - Scratch pad checksum invalid!
[12:39:38][D][sensor:093]: 'Temperature #2': Sending state nan °C with 0 decimals of accuracy
[12:39:38][W][dallas.sensor:156]: 'Temperature #3' - Scratch pad checksum invalid!
[12:39:38][D][sensor:093]: 'Temperature #3': Sending state nan °C with 0 decimals of accuracy
```

ldeffenb, 2023-10-02:
I solved this finally.   Apparently the "0x" prefix that is showing up on the "Found sensors:" listing is a red herring.  The displayed value is in fact decimal.   If you drop the 0x when you copy it into the sensor definition, it works perfectly fine!

closed 2023-10-02 by ldeffenb

MarkoPaasila, 2023-10-03:
Did you figure out this on your own, or is it discussed elsewhere? I've been having scratchpad errors with several components and haven't found a solution yet. Something changed about half a year ago and I haven't been able to upgrade since. I'll give this a try.