persuader72/esphome-components #17: Hi, I get "while parsing a block mapping" all the time, what am I doing wrong?

Bundas77, 2024-01-23:
INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/shely.yaml...
ERROR Error while reading config: Invalid YAML syntax:

while parsing a block mapping
  in "/config/esphome/shely.yaml", line 65, column 3:
      pin_a: GPIO3
      ^
expected <block end>, but found '-'
  in "/config/esphome/shely.yaml", line 68, column 3:
      - platform: shelly_dallas
      ^

`# Basic config

substitutions:
  device_name: Shelly1

esphome:
  name: shelly_detached
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name} Hotspot
    password: ""

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

# Enable OTA updates
ota:

# Enable Web server (optional).
web_server:
  port: 80

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: ${device_name} ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: ${device_name} IP
    ssid:
      name: ${device_name} SSID
    bssid:
      name: ${device_name} BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${device_name} Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${device_name} WiFi Signal
    update_interval: 60s

shelly_dallas:
  pin_a: GPIO3
  pin_b: GPIO0

  - platform: shelly_dallas
    address: 0
    name: "Temperature #1"
  - platform: shelly_dallas
    address: 0
    name: "Temperature #2"
  - platform: shelly_dallas
    address: 0
    name: "Temperature #3"

# Shelly 1 detached switch config with multi click options and fallback in case of wifi or api fail

switch:
  - platform: gpio
    name: shelly1 ${device_name}
    pin: GPIO4
    id: shelly_relay
    restore_mode: restore_default_off

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO5
    name: ${device_name} button
    # config for multi click actions
    on_multi_click:
      # double click
      - timing:
          - ON for at most 1s
          - OFF for at most 1s
          - ON for at most 1s
          - OFF for at least 0.2s
        then:
          - if:
              condition:
                and:
                  - wifi.connected:
                  - api.connected:
              # send double click event in case wifi and api are conncected
              then:
                - homeassistant.event:
                    event: esphome.button_pressed
                    data:
                      title: shelly ${device_name} double click
              # toggle relay in case either wifi or api are not connected
              else:
                - switch.toggle: shelly_relay
        # long click
      - timing:
          - ON for at least 1.5s
        then:
          - if:
              condition:
                and:
                  - wifi.connected:
                  - api.connected:
              # send long click event in case wifi and api are conncected
              then:
                - homeassistant.event:
                    event: esphome.button_pressed
                    data:
                      title: shelly ${device_name} long click
              # toggle relay in case either wifi or api are not connected
              else:
                - switch.toggle: shelly_relay
        # single click
      - timing:
          - ON for at most 1s
          - OFF for at least 0.5s
        then:
          - if:
              condition:
                and:
                  - wifi.connected:
                  - api.connected:
              # send single click event in case wifi and api are conncected
              then:
                - homeassistant.event:
                    event: esphome.button_pressed
                    data:
                      title: shelly ${device_name} short click
              # toggle relay in case either wifi or api are not connected
              else:
                - switch.toggle: shelly_relay
    internal: true
    id: button

`

MarkoPaasila, 2024-01-23:
Maybe check indentation?

MarkoPaasila, 2024-01-24:
Try addin a single component at a time.

Bundas77, 2024-01-24:
[image]

Bundas77, 2024-01-24:
[image]

Bundas77, 2024-01-24:
> Try addin a single component at a time.

That's what I tried, then it do not find the component ...

Bundas77, 2024-01-24:
[image]

persuader72, 2024-01-28:
Hi to all sorry for late reply

Maybe this custom component is a little outdated, this week i will try to
update it to one more recent version of ephome...

Which version are you currently using?

Stefano

Bundas77, 2024-01-31:
Latest 2023.12.9

Bundas77, 2024-02-12:
Hi!
Any progress? 
András

persuader72, 2024-05-16:
Hi to all at the end I was able to remove my shelly from the wall and do some debug on it.
Now the component is updated to last esphome e is working for me.
The component is also now compatible with the new external_components configurations options.

There is an example in the main directory