persuader72/esphome-components #19: No Sensors Found (2024.2.2)

m1k3f15h, 2024-07-29:
Hey Guys,

Does Anyone have the same Problem?
I tryed with 2 different Sensors...
Shelly Plus 2 PM + Addon

[07:25:30][I][app:102]: ESPHome version 2024.2.2 compiled on Jul 29 2024, 07:04:39
[07:25:31][C][dallas.sensor:077]: DallasComponent:
[07:25:31][C][dallas.sensor:078]:   Pin: GPIO3
[07:25:31][C][dallas.sensor:079]:   Update Interval: 60.0s
[07:25:31][W][dallas.sensor:082]:   Found no sensors!

Greetings from Austria
Mike

persuader72, 2024-07-29:
Hi, I never tested this component with Shelly 2PM maybe the addon of this device uses differents gpio respect the shelly 1 addon.

Can you please send me the link to the product  web page of your addon? I will try to understand what changes it need to be made to make it works.

Regards
Stefano

m1k3f15h, 2024-07-29:
Hello @persuader72 

I tested it with GPIO1 instead of GPIO3 ...
and it worked :)

so thank you for your response.
maybe you could write it in your Doku :)

dears Mike

closed 2024-07-29 by m1k3f15h

rici44, 2024-10-12:
Hi @persuader72  Is your custom_component for shelly add-on working also with Shelly Plus 1PM? I tested it with dht and dallas sensors, setup pin_a: GPIO3 pin_b: GPIO0, but now sensors were found. I also tried pin_a: GPIO19 [as shown here](https://devices.esphome.io/devices/Shelly-Plus-1PM#gpio-pinout) but without success. What gpio pins should be used for Shelly Plus 1PM, Shelly Plus 2PM or Shelly Plus 1?

rici44, 2024-10-12:
Update: 
Shelly Plus 1PM with DHT22 is working with this pinout: 
pin_a: GPIO1 
pin_b: GPIO0