persuader72/esphome-components #22: Link to device broken

Mincol, 2024-10-05:
I believe that link in description is broken.

A component to support [Temperature Sensor AddOn for Shelly 1/1PM](https://shop.shelly.cloud/temperature-sensor-addon-for-shelly-1-1pm-wifi-smart-home-automation#312),

Should it be https://www.shelly.com/en/products/shop/shelly-plus-add-on ?

klim123123, 2024-10-05:
Temperature Sensor AddOn for Shelly 1/1PM is not the same as Shelly Plus Add-On
Correct link is https://shelly.digital/product/temperature-sensor-addon-for-shelly-1-1pm/

Mincol, 2024-10-06:
Thanks. So my search for support of "Shelly Plus Add-on" in ESPhome continue...

martmaiste, 2025-02-23:
Thanks!

Also for Shelly Plus 1PM

#pin_a: GPIO1 #For Shelly Plus 2PM