persuader72/esphome-components #24: DTH sensor does not work with shelly plus addon

Delphius7, 2025-09-09:
First of all thanks for this component and your effort!
I have recently changed the firmware on one of my `Shelly Plus 1` devices to ESPhome (**2025.8.3**). I have a `Shelly Plus Addon` connected with a DHT21 and a reed sensor.
The DHT21 sensor was working well with the original shelly firmware so I am sure it is working and also cabled correctly. However i just cannot get it to work with this component.

Does anyone have an idea what i am doing wrong?
Is it even using the shelly_dht platform? in the logs it shows dht?
Thanks in advance!

Here is my esphome yaml:
```
sensor:
  - platform: shelly_dht
    pin_a: GPIO1
    pin_b: GPIO0
    model: DHT22
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    update_interval: 60s
```

and here the important parts of the DEBUG log:
```
[00:21:09][C][dht:021]: DHT:
[00:21:09][C][dht:022]:   PinA: GPIO1
[00:21:09][C][dht:023]:   PinB: GPIO0
[00:21:09][C][dht:029]:   Model: DHT22 (or equivalent)
[00:21:09][C][dht:032]:   Update Interval: 60.0s
[00:21:09][C][dht:034]:   Temperature 'Temperature'
[00:21:09][C][dht:034]:     State Class: 'measurement'
[00:21:09][C][dht:034]:     Unit of Measurement: '°C'
[00:21:09][C][dht:034]:     Accuracy Decimals: 1
[00:21:09][C][dht:034]:     Device Class: 'temperature'
[00:21:09][C][dht:035]:   Humidity 'Humidity'
[00:21:09][C][dht:035]:     State Class: 'measurement'
[00:21:09][C][dht:035]:     Unit of Measurement: '%'
[00:21:09][C][dht:035]:     Accuracy Decimals: 0
[00:21:09][C][dht:035]:     Device Class: 'humidity'
.....
[00:22:08][W][dht:175]: Requesting data from DHT failed!
[00:22:08][W][dht:065]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[00:22:08][D][sensor:103]: 'Temperature': Sending state nan °C with 1 decimals of accuracy
[00:22:08][D][sensor:103]: 'Humidity': Sending state nan % with 0 decimals of accuracy
```

persuader72, 2025-09-10:
Thanks,

Your configuration look correct,  Maybe there is a compatibility problem of this component with the last versions Esphome I can't test it right now because I lost my DHT :).

I will buy a new one and debug it in the next days,.. I will keep you in touch.

persuader72, 2025-09-21:
H,i my DHT sensor just arrived and It works with the last version of esphome... Can you share your full config to do a crosscheck?